sanni/cartreader #1088: Can't write to Satellaview 8M Memory Pack

ChronoMoogle, 2025-04-12:
### OSCR firmware version

14.3

### OSCR hardware version

HW5 Rev6

### Attach OSCR_LOG.txt file here

[OSCR_LOG.txt](https://github.com/user-attachments/files/19719459/OSCR_LOG.txt)

### System used

SNES/Super Famicom

### Describe the issue.

I have an issue when trying to write to Satellaview 8M Memory Cartridges:
The progress won't go beyond the "Resetting" step. 
Cartridges are well connected and can be dumped with good checksum, also if I remove the 8M during the currently infinite "Resetting"-step, the progress will detect the cartridge being removed and stop the progress, so its not frozen. I powercycled the BS-X cartridge multiple times between retries.

sanni, 2025-04-15:
According to your log file you successfully flashed a 8M memory pack once or twice and then later on it did not work anymore as it failed to erase the cartridge. 

Do you have multiple 8M packs or is it the same one that worked before and now does not?

Maybe dump the 8M pack in the SV menu and then open the file in an hex editor to see what part of the pack was not erased, maybe this will give a clue to why it fails.

Maybe related: There was an issue with flashing the 8M memory packs recently where the clock generator was the culprit. The user in that case could also not calibrate the clock generator like described here: https://github.com/sanni/cartreader/wiki/Calibrating-the-ClockGen
Maybe something similar is happening here. If this is the case you can get a good quality clock gen here: https://www.adafruit.com/product/2045

On a side note: You should right-click on all the database txt files and mark them as hidden so they don't appear in the file browser of the OSCR.

sanni, 2025-04-22:
Ok it fails to fully erase the memory pack, no pattern to see, probably just an error in the erase function that only happens with some carts. According to [nocash's site](https://www.problemkaputt.de/fullsnes.htm#snescartdatapackslotssatellaviewlikeminicartridgeslot), four different types exist one of which does not support chip erase and need sector erase instead. Could also just be a timing issue.
So I guess the first thing to try would be to implement sector erase instead of chip erase. Sadly I sold my Satellaview stuff a long time ago.